meshtastic/Meshtastic-gui-installer #32: Detect Ports does not work on (my) Windows System

caveman99, 2022-02-01:
If needed, i can attach the output of `powershell.exe "Get-PnpDevice -PresentOnly | Format-List"` too...

```
PS C:\Users\thomas> meshtastic-flasher.exe
Traceback (most recent call last):
  File "C:\Users\thomas\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic_flasher\installer.py", line 283, in detect
    supported_devices_detected = detect_supported_devices()
  File "C:\Users\thomas\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\util.py", line 300, in detect_supported_devices
    _, sp_output = subprocess.getstatusoutput('powershell.exe "Get-PnpDevice -PresentOnly | Format-List"')
  File "C:\Users\thomas\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 624, in getstatusoutput
    data = check_output(cmd, shell=True, text=True, stderr=STDOUT)
  File "C:\Users\thomas\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 420, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "C:\Users\thomas\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 503, in run
    stdout, stderr = process.communicate(input, timeout=timeout)
  File "C:\Users\thomas\AppData\Local\Programs\Python\Python310\lib\subprocess.py", line 1136, in communicate
    stdout = self.stdout.read()
  File "C:\Users\thomas\AppData\Local\Programs\Python\Python310\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 12273: character maps to <undefined>
```

mkinney, 2022-02-01:
After doing a bit of research, I'm thinking about adding a snippet in front of the command:

```
powershell.exe "$PSDefaultParameterValues['Out-File:Encoding'] = 'utf8'; Get-PnpDevice -PresentOnly"
```

mkinney, 2022-02-01:
Or maybe...

```
powershell.exe  /c chcp 65001 "Get-PnpDevice -PresentOnly"
```

closed 2022-02-01 by mkinney

mkinney, 2022-02-02:
Can you try the new release and let me know if you still have problems?

caveman99, 2022-02-02:
The described issue is gone. There are still a few corner cases but nothing related to this bug! Thanks!

PR #33 (2022-02-01): change windows encoding
Fix #32