meshtastic/Meshtastic-gui-installer #39: `pip` installation on MacOS

jfirwin, 2022-02-02:
I've been unable to get this running on a MacOS machine running Catalina (10.15.4). I was able to successfully get it running on a clean installation of Monterey, so it could be my python environment potentially. Here are the errors.

Installation method: 
```
pip3 install meshtastic-flasher
```
Results:
```
$ meshtastic-flasher
Traceback (most recent call last):
  File "/usr/local/bin/meshtastic-flasher", line 5, in <module>
    from meshtastic_flasher.installer import main
  File "/usr/local/lib/python3.9/site-packages/meshtastic_flasher/installer.py", line 19, in <module>
    from meshtastic.util import detect_supported_devices, findPorts, detect_windows_needs_driver
ImportError: cannot import name 'detect_supported_devices' from 'meshtastic.util' (/usr/local/lib/python3.9/site-packages/meshtastic/util.py

$ python3 --version
Python 3.9.2
```
Upgraded:
```
pip3 install --upgrade meshtastic-flasher
```
Upgraded Results (Same as above): 
```
$ meshtastic-flasher
Traceback (most recent call last):
  File "/usr/local/bin/meshtastic-flasher", line 5, in <module>
    from meshtastic_flasher.installer import main
  File "/usr/local/lib/python3.9/site-packages/meshtastic_flasher/installer.py", line 19, in <module>
    from meshtastic.util import detect_supported_devices, findPorts, detect_windows_needs_driver
ImportError: cannot import name 'detect_supported_devices' from 'meshtastic.util' (/usr/local/lib/python3.9/site-packages/meshtastic/util.py)
```

caveman99, 2022-02-02:
try forcing a `pip3 install --upgrade meshtastic`

jfirwin, 2022-02-02:
That was it. I was on `meshtastic 1.2.76`, and upgraded to `meshtastic 1.2.80` and it's working now.

jfirwin, 2022-02-02:
I'm closing this issue, but we will probably want to note something about meshtastic-python being up to date in the documentation.

closed 2022-02-02 by jfirwin

caveman99, 2022-02-02:
the dependancy to the api lib needs to be pinned to a minimal version. Obviously on the standalone binaries tht's always the case since they're built from scratch.