meshtastic/Meshtastic-gui-installer #52: raspberry pi arm64 research

mkinney, 2022-02-07:
Can we get `meshtastic-flasher` working on a raspberry pi?

PySide6 nor PySide2 is not available.

See https://bugreports.qt.io/browse/PYSIDE-802?focusedCommentId=418104&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel

mkinney, 2022-02-07:
This seems to work:

```
Raspbian Buster has now been released and Pyside2 can be installed using:

$apt-get install python3-pyside2.qt3dcore python3-pyside2.qt3dinput python3-pyside2.qt3dlogic python3-pyside2.qt3drender python3-pyside2.qtcharts python3-pyside2.qtconcurrent python3-pyside2.qtcore python3-pyside2.qtgui python3-pyside2.qthelp python3-pyside2.qtlocation python3-pyside2.qtmultimedia python3-pyside2.qtmultimediawidgets python3-pyside2.qtnetwork python3-pyside2.qtopengl python3-pyside2.qtpositioning python3-pyside2.qtprintsupport python3-pyside2.qtqml python3-pyside2.qtquick python3-pyside2.qtquickwidgets python3-pyside2.qtscript python3-pyside2.qtscripttools python3-pyside2.qtsensors python3-pyside2.qtsql python3-pyside2.qtsvg python3-pyside2.qttest python3-pyside2.qttexttospeech python3-pyside2.qtuitools python3-pyside2.qtwebchannel python3-pyside2.qtwebsockets python3-pyside2.qtwidgets python3-pyside2.qtx11extras python3-pyside2.qtxml python3-pyside2.qtxmlpatterns 
```

How would the imports need to look for a simple hello_world.py?

mkinney, 2022-02-07:
Could setup a cross compiling environment.

Tried: https://medium.com/@stonepreston/how-to-cross-compile-a-cmake-c-application-for-the-raspberry-pi-4-on-ubuntu-20-04-bac6735d36df

and

https://ilyas-hamadouche.medium.com/creating-a-cross-platform-toolchain-for-raspberry-pi-4-5c626d908b9d

both errored

This looks promising:

https://tttapa.github.io/Pages/Raspberry-Pi/C++-Development/Building-The-Toolchain.html

mkinney, 2022-02-07:
From: https://github.com/piwheels/packages/issues/4#issuecomment-772058821

[build_pyside2.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8012705/build_pyside2.txt)

[install_qt5.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8012706/install_qt5.txt)

dfloer, 2022-02-07:
I tested and confirmed that `pip install meshtastic-flasher` works* on ArchlinuxARM.

Installation on a system was essentially:
```bash
pacman -S qt6 pyside6
pip install meshtastic-flasher
```
I had the entire qt6 group installed, and a more targetted install of only the required dependencies may be desired. There may be other system dependencies I missed here.

*Currently, there is a version mismatch between `pyside6` and `shiboken6` in the armv7h repos only, which prevents pip from ever being able to resolve the dependencies to install it. I built the updated package myself to confirm that this is indeed the problem. The aarch64 repos have the proper versions of both packages, so this shouldn't be an issue there.