meshtastic/Meshtastic-gui-installer #63: need to create/update partitions

mkinney, 2022-02-10:
See https://github.com/meshtastic/Meshtastic-device/blob/master/bin/genpartitions.py
and https://demo-dijiudu.readthedocs.io/en/latest/api-guides/partition-tables.html

@mc-hamster Will share details on what to write and when.

mkinney, 2022-02-10:
https://meshtastic.discourse.group/t/ttgo-t-beam-partitions/5028

mc-hamster, 2022-02-10:
let's gather data before starting work on this. may be pebkac.

mkinney, 2022-02-13:
potato potato

mkinney, 2022-02-13:
I should say that it was not meant in a derogatory or condescending manner... just short-hand "user error".

mkinney, 2022-02-13:
But, if we need to change some code or something, please share.

ElectricRoxy, 2022-02-13:
Well, the issue was I believed the documentation. That was wrong, not me.

mkinney, 2022-02-13:
I've not been paying attention to the underlying issue.

Does the documentation need to be changed?

ElectricRoxy, 2022-02-13:
it has been updated -
CAUTION
It has been reported that some of this information is out of date.
FIXME - Investigate and rewrite document to reflect the current ESP32 Partition mitigation.

mkinney, 2022-02-25:
I'm going to close this issue as there does not seem to be anything to do at this point. Feel free to re-open this issue (or create a new issue) when we know more about what needs to be changed/updated.

closed 2022-02-25 by mkinney