meshtastic/Meshtastic-gui-installer #67: add clickable links to message boxes

mkinney, 2022-02-11:
Would be nice to have clickable links in message boxes to pages like Meshtastic docs and/or vendor docs

ex:
https://docs.rakwireless.com/assets/images/wisblock/rak11200/quickstart/rak11200-Boot0-for-flashing.png

ElectricRoxy, 2022-02-13:
Give me a few days to get my git skills up and I'll take this on?

mkinney, 2022-02-14:
Let me know if you need help with git.

Recently, made the labels “Version” and “Device” clickable links, if that helps.

Even a snippet or small sample code would be helpful.

mkinney, 2022-02-17:
I *finally* figured it out:

```
        msg_box = QMessageBox()
        msg_box.setTextFormat(QtCore.Qt.RichText)  # this is what makes the links clickable
        msg_box.setText("Tips:<br><br>"
                        "If having issues flashing the device, be sure there is only one device connected "
                        "and no other applications are using that communications port.<br><br>"
                        "If still having problems, unplug the device, then re-plugin the device.<br><br>"
                        "If still having problems, may need to install driver. "
                        "See <a href='https://meshtastic.org/docs/getting-started/flashing-esp32'>Flashing</a> for more info.<br><br>"
                        "If still having problems, try rebooting the pc.<br><br>"
                        "If you get a 'Critical Fault #6' on a T-Beam, it probably means you need to use "
                        "v1.1 or v2.1.1.6 firmware.")
        msg_box.setStandardButtons(QMessageBox.Ok)
        msg_box.exec()
```

Do you need help getting going with git?

ElectricRoxy, 2022-02-17:
I've got it installed working, watched a few tutorials on the tube of U . I think after this weekend I'll have some more time to get going with you. Software release week at work.