meshtastic/Meshtastic-gui-installer #102: Application closes without notice if Settings clicked without having any devices connected.

fsgl1tch, 2022-02-21:
Either the clicking the Settings without devices connected shows a warning or the settings menu could be disabled till a device is connected (greyed out). I would say that disabled button seems the most reasonable. Design decision time!

mkinney, 2022-02-21:
It is a catch-22. I don't really know if there is a device connected until the Settings button is pressed.

On the prior version 1.0.71, it would just simply close m-flasher if there were no devices connected. In the latest version, it shows a dialog that the user needs to plugin a device.

mkinney, 2022-02-21:
If you're running from a cloned repo, you may need to:

```
git checkout master
git pull
pip install .
meshtastic-flasher
```

fsgl1tch, 2022-02-22:
Yes, it seems that in newer versions the Settings for no device is handled more gracefully!

closed 2022-02-22 by fsgl1tch