meshtastic/Meshtastic-gui-installer #118: Change out QMessageBox calls

mkinney, 2022-02-24:
A QMessageBox.information or .warning will not show a title in the WindowTitle on mac. For example:

[image]

but the window title does work on a normal QDialog on mac:
[image]

So, we should change out the QMessageBox calls with our own class derived from QDialog.

mkinney, 2022-02-24:
Filed issue https://bugreports.qt.io/browse/PYSIDE-1833

pratham-saraf, 2022-02-25:
hey @mkinney 
i would like to know regarding status of this issue
and if its still open i would like to try it  as my first contributuion.
Also can you please point me from where to start regarding this
regards

mkinney, 2022-02-25:
Great!

Yes, this is still an open issue.

I was hoping to add some flag or option that would enable the title to be seen, but apparently it is "by design" on mac.

The way I'd approach it would be to create a new form CustomMessageBox() in a new file custom_message_box.py that would derive from QDialog (much like most of the other forms). Then I'd change one QMessageBox.information() call to it and ensure it works as expected. Then I'd change a .warning and .question. If those look/work right on all three OSs, I'd change all of the 40 calls in the code.

You don't have to do all of it at once. Just creating the new form and swapping out one of the calls would be a good first step. A good first one to replace might be the "Device Settings" on the main settings.py page. https://github.com/meshtastic/Meshtastic-gui-installer/blob/master/meshtastic_flasher/settings.py#L185 For that, you don't need a radio connected.

Let me know if you have questions or want some help getting started.

pratham-saraf, 2022-02-26:
@mkinney  
Thanks for the response 
As it is my first time dealing with Qmessage i would first have to look into its documentation.
Anyother helpful suggestion from your end which might help me regarding this
regards