meshtastic/Meshtastic-gui-installer #123: user reports that "The GUI doesn’t work in high Sierra"

mkinney, 2022-02-28:
GrimReaper user on Discord reported that the GUI doesn’t work in high Sierra.

mkinney, 2022-03-07:
Not sure we want to support OS's this old.

mkinney, 2022-03-07:
[image]

mkinney, 2022-03-30:
Per: https://docs.github.com/en/actions/using-github-hosted-runners/about-github-hosted-runners 

[image]

May consider using  macOS Catalina 10.15 or `macos-10.15` for the mac release runner.

garthvh, 2022-07-04:
Expected

closed 2022-07-04 by garthvh