meshtastic/Meshtastic-gui-installer #130: user reports "detected" text is garbled

mkinney, 2022-03-04:
user reports "detected" text is garbled

The white color txt in Left side-middle about the installed FW version, is garbled, out of focus, not one line txt words is on top of other words.

https://meshtastic.discourse.group/t/meshtastic-gui-installer-flasher-big-thank-you/5232

I suspect win10 is the OS

geeksville, 2022-03-04:
This issue has been mentioned on **Meshtastic**. There might be relevant details there:

https://meshtastic.discourse.group/t/meshtastic-gui-installer-flasher-big-thank-you/5232/3

mkinney, 2022-03-06:
[image]

mkinney, 2022-03-06:
[image]

closed 2022-03-06 by mkinney

mkinney, 2022-03-06:
https://github.com/meshtastic/Meshtastic-gui-installer/releases/tag/1.0.90

PR #131 (2022-03-06): move detected text so it does not display over other text
Fix: #130