meshtastic/Meshtastic-gui-installer #140: Having a different issue flashing T-Beam

ridgebackdad, 2022-03-12:
Had the same issue as #94  - but worked around that by deleting the images and pulling them down again. But now, I'm having this issue:

`Step 1/4 esp32 full
ESPTOOL Using command:--port /dev/ttyACM0 erase_flash
esptool.py v3.2
Serial port /dev/ttyACM0
Connecting...

.

.

.

Detecting chip type...

Unsupported detection protocol, switching and trying again...
Connecting...

.

.

.

Detecting chip type...

ESP32

Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 44:17:93:87:84:10
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 13.0s
Hard resetting via RTS pin...
Step 2/4 esp32 full
ESPTOOL Using command:--port /dev/ttyACM0 write_flash 0x1000 1.2.57.f7c6955/system-info.bin
esptool.py v3.2
Serial port /dev/ttyACM0
Connecting...

.

.

.

.

Detecting chip type...

Unsupported detection protocol, switching and trying again...
Connecting...

.

Detecting chip type...

ESP32

Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 44:17:93:87:84:10
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x0000ffff...
Compressed 61440 bytes to 11950...
Writing at 0x00001000... (100 %)
Wrote 61440 bytes (11950 compressed) at 0x00001000 in 1.6 seconds (effective 299.8 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...
Step 3/4 esp32 full
ESPTOOL Using command:--port /dev/ttyACM0 write_flash 0x00390000 1.2.57.f7c6955/spiffs-1.2.57.f7c6955.bin
esptool.py v3.2
Serial port /dev/ttyACM0
Connecting...

.

.

.

Detecting chip type...

Unsupported detection protocol, switching and trying again...
Connecting...

.

Detecting chip type...

ESP32

Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 44:17:93:87:84:10
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00390000 to 0x003fffff...
Compressed 458752 bytes to 273911...
Writing at 0x00390000... (5 %)
Writing at 0x003962f7... (11 %)
Writing at 0x0039bbd2... (17 %)
Writing at 0x003a187a... (23 %)
Writing at 0x003a772a... (29 %)
Writing at 0x003ad35c... (35 %)
Writing at 0x003b727a... (41 %)
Writing at 0x003bdfd2... (47 %)
Writing at 0x003c4d67... (52 %)
Writing at 0x003cbd6a... (58 %)
Writing at 0x003d2b41... (64 %)
Writing at 0x003d997d... (70 %)
Writing at 0x003e0725... (76 %)
Writing at 0x003e758f... (82 %)
Writing at 0x003ee20a... (88 %)
Writing at 0x003f4d30... (94 %)
Writing at 0x003fbb09... (100 %)
Wrote 458752 bytes (273911 compressed) at 0x00390000 in 24.3 seconds (effective 151.3 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting via RTS pin...
Step 4/4 esp32 full
ESPTOOL Using command:--port /dev/ttyACM0 write_flash 0x10000 1.2.57.f7c6955/firmware-tlora-v2-1-1.2.57.f7c6955.bin
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '1.2.57.f7c6955/firmware-tlora-v2-1-1.2.57.f7c6955.bin'
`
It then just sits there.  I have exited the flasher app, but if I try it again, I can occasionally get a core dump.

ridgebackdad, 2022-03-12:
Looks like core dump is a little more common than I had thought.  Tried again and it core dumps again:

`~/Projects/meshtastic/venv  meshtastic-flasher
pypi_version:1.0.98
running: 1.0.98
start of detect
group_to_search:dialout
Detected T-Lora
Detected T-Beam
Detected T-Beam
Detected T-Beam
Detected T-Beam
Detected T-Lora
Detected T-Lora
User said no to device having Meshtastic on the device so no detection using the Meshtastic python method
filenames:['1.2.57.f7c6955/firmware-tbeam0.7-1.2.57.f7c6955.bin', '1.2.57.f7c6955/firmware-tlora-v2-1-1.6-1.2.57.f7c6955.bin', '1.2.57.f7c6955/firmware-tlora-v2-1.2.57.f7c6955.bin', '1.2.57.f7c6955/firmware-tlora-v1-1.2.57.f7c6955.bin', '1.2.57.f7c6955/firmware-heltec-v2.1-1.2.57.f7c6955.bin', '1.2.57.f7c6955/firmware-meshtastic-diy-v1-1.2.57.f7c6955.bin', '1.2.57.f7c6955/firmware-heltec-v2.0-1.2.57.f7c6955.bin', '1.2.57.f7c6955/firmware-heltec-v1-1.2.57.f7c6955.bin', '1.2.57.f7c6955/firmware-tlora_v1_3-1.2.57.f7c6955.bin', '1.2.57.f7c6955/firmware-tbeam-1.2.57.f7c6955.bin', '1.2.57.f7c6955/firmware-rak11200-1.2.57.f7c6955.bin']
end of detect
Flash was clicked.
update only is not checked
User confirmed they want to flash
do esptool update_only:False
QBasicTimer::stop: Failed. Possibly trying to stop from a different thread
[1]    8567 segmentation fault (core dumped)  meshtastic-flasher
`

mkinney, 2022-03-13:
How was this installed? Pip or the MacApp? What version of OS?

It is weird that it says, "firmware-tlora-v2-1-1.2.57.f7c6955.bin" in the first flashing, but you had t-beam selected?

ridgebackdad, 2022-03-13:
Trying to flash it from Ubuntu Linux (20.04 LTS).  Installed software via pip.

Same issue over and over.  Rebooted the Linux box - no joy. Removed the software, re-installed it - and it worked fine.

Issue resolved.

closed 2022-03-13 by ridgebackdad