meshtastic/Meshtastic-gui-installer #141: Cannot download meshtastic-flasher via pip due to "Trojan" virus warning

badgerwhisperer, 2022-03-13:
I attempted to download the meshtastic-flasher through the link under the "Assets" section of page https://github.com/meshtastic/Meshtastic-gui-installer/releases/tag/winapp1.0.0. My virus protection automatically stopped the download and said it detected a Trojan virus. I am not comfortable downloading everything manually yet. Is there another link to download the flasher through? Below is the url to the link that supposedly had the Trojan in it. Please beware. 

https://github.com/meshtastic/Meshtastic-gui-installer/releases/download/winapp1.0.0/meshtastic-flasher.zip

andrekir, 2022-03-13:
what virus protection software are you using?

badgerwhisperer, 2022-03-13:
Whatever comes standard with hp chromebooks.

garthvh, 2022-03-13:
This is a windows defender alert, are you sure it is a chromebook you are running? I have the file being analyzed by Microsoft, it appears to be a machine learning false positive bit I probably won't know for sure until this week.

badgerwhisperer, 2022-03-13:
I apologize, it is not a chromebook. Just a regular 64-bit HP laptop. Thank you for sending the file in to be analyzed. Any suggestions on how to work around a false positive?

garthvh, 2022-03-13:
> I apologize, it is not a chromebook. Just a regular 64-bit HP laptop. Thank you for sending the file in to be analyzed. Any suggestions on how to work around a false positive?

I submitted it on Friday, so we are well within the timeframe for Microsoft to review it, I would use the flasher from pip for now while we await a determination.

closed 2022-03-14 by garthvh