meshtastic/Meshtastic-gui-installer #142: "Resource busy" when attempting to flash

raldi, 2022-03-13:
I get the following exception when I attempt to flash my new T-Beam even though `sudo lsof /dev/cu.usbmodem54240349111` has no output:

```
Step 1/4 esp32 full
ESPTOOL Using command:--port /dev/cu.usbmodem54240349111 erase_flash
esptool.py v3.2
Serial port /dev/cu.usbmodem54240349111
Traceback (most recent call last):
File "/Users/raldi/meshtastic-flasher/venv/lib/python3.9/site-packages/serial/serialposix.py", line 322, in open

self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
OSError
:
[Errno 16] Resource busy: '/dev/cu.usbmodem54240349111'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/Users/raldi/meshtastic-flasher/venv/lib/python3.9/site-packages/meshtastic_flasher/esptool_form.py", line 70, in run

esptool.main(command)
File "/Users/raldi/meshtastic-flasher/venv/bin/esptool.py", line 4541, in main

esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
File "/Users/raldi/meshtastic-flasher/venv/bin/esptool.py", line 121, in get_default_connected_device

_esp = ESPLoader.detect_chip(each_port, initial_baud, before, trace,
File "/Users/raldi/meshtastic-flasher/venv/bin/esptool.py", line 372, in detect_chip

detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
File "/Users/raldi/meshtastic-flasher/venv/bin/esptool.py", line 329, in __init__

self._port = serial.serial_for_url(port)
File "/Users/raldi/meshtastic-flasher/venv/lib/python3.9/site-packages/serial/__init__.py", line 90, in serial_for_url

instance.open()
File "/Users/raldi/meshtastic-flasher/venv/lib/python3.9/site-packages/serial/serialposix.py", line 325, in open

raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil
.
SerialException
:
[Errno 16] could not open port /dev/cu.usbmodem54240349111: [Errno 16] Resource busy: '/dev/cu.usbmodem54240349111'
```

I'll debug and post my solution to this thread when I find it.

raldi, 2022-03-13:
Looks like maybe /dev/cu.usbmodem54240349111 was a leftover, unrelated device that was plugged into my machine before the T-Beam; after rebooting, it's gone. 

Unfortunately, it appears the T-Beam isn't getting a device at all; `ls /dev` has the same output when it's connected as when it's not. I've installed both the CP210X and CH9102 drivers, and rebooted, and they seem to be installed: https://i.imgur.com/WR8GF21.png

raldi, 2022-03-13:
Noticed the instructions say, "With the latest versions of MacOS, the USB Serial driver is built-in. Do NOT download the USB device drivers unless required. If you downloaded/installed any already, please remove them", so I ran `sudo rm -rf /Library/Extensions/usbserial.kext` and rebooted.

Now I get a `/dev/cu.usbmodem54240349111` and `/dev/tty.usbmodem54240349111` when I plug in the T-Beam, and my System Report shows a bit more detail under the USB device:

```
USB Single Serial:

  Product ID:	0x55d4
  Vendor ID:	0x1a86
  Version:	4.43
  Serial Number:	5424034911
  Speed:	Up to 12 Mb/sec
  Location ID:	0x14210000 / 3
  Current Available (mA):	500
  Current Required (mA):	134
  Extra Operating Current (mA):	0
```

...but I still get the same exception when I attempt to flash.

raldi, 2022-03-13:
Hmm, maybe I need both drivers after all. The device I have is a https://www.aliexpress.com/item/4001178678568.html ("LILYGO® TTGO Meshtastic T-Beam V1.1 ESP32 LoRa 433/868/915/923Mhz Wireless Module WiFi GPS NEO-6M With OLED Display for Arduino") and the ESP32 section of the Meshtastic docs seems to say I need both.

raldi, 2022-03-13:
Based on https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/issues/139 , I tried switching to https://github.com/Xinyuan-LilyGO/LilyGo-T-Call-SIM800/files/7037260/CH9102_Mac_Driver.zip and I'm able to get a lot further:

`Step 1/4 esp32 full
ESPTOOL Using command:--port /dev/cu.usbmodem54240349111 erase_flash
esptool.py v3.2
Serial port /dev/cu.usbmodem54240349111
Connecting...

.

Detecting chip type...

Unsupported detection protocol, switching and trying again...
Connecting...

.

Detecting chip type...

ESP32

Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 58:bf:25:04:cf:10
Uploading stub...
Traceback (most recent call last):
File "/Users/raldi/meshtastic-flasher/venv/lib/python3.9/site-packages/meshtastic_flasher/esptool_form.py", line 70, in run

esptool.main(command)
File "/Users/raldi/meshtastic-flasher/venv/bin/esptool.py", line 4564, in main

esp = esp.run_stub()
File "/Users/raldi/meshtastic-flasher/venv/bin/esptool.py", line 879, in run_stub

self.mem_block(stub[field][from_offs:to_offs], seq)
File "/Users/raldi/meshtastic-flasher/venv/bin/esptool.py", line 752, in mem_block

return self.check_command("write to target RAM", self.ESP_MEM_DATA,
File "/Users/raldi/meshtastic-flasher/venv/bin/esptool.py", line 505, in check_command

raise FatalError.WithResult('Failed to %s' % op_description, status_bytes)
esptool
.
FatalError
:
Failed to write to target RAM (result was 01070000: Operation timed out)`

mkinney, 2022-03-14:
Sorry you are having so many problems. I think you might need to install the CH9102 driver (see https://meshtastic.org/docs/getting-started/flashing-firmware under macOS

We tend to have people on the Discord channel that might be able to help you more interactively, if you're willing to join Discord and ask questions there. Although, I do appreciate your opening an issue to track any issues.

I would like to track down any changes to the documentation we might need to make, too.

mkinney, 2022-03-17:
I wonder the the LilyGo drivers are causing the issues. The Meshtastic docs seem to refer to different drivers: 
https://meshtastic.org/docs/getting-started/flashing-firmware#install-appropriate-drivers

closed 2022-07-04 by garthvh