meshtastic/Meshtastic-gui-installer #144: Windows meshtastic-flasher says "The system cannot find the path specified." and exits

raldi, 2022-03-13:
When I try to run the Windows version of meshtastic-flasher, it just pops up a terminal window saying `The system cannot find the path specified.` and then it disappears and the program exits.

mkinney, 2022-03-14:
It is probably related to `python3`. The version 1.0.0 of WinApp assumes that python3 is installed.

The next version of WinApp will add a bit more checking.

Can you open a command prompt and run: `python --version`?

Do you have python3 installed? If so, how was it installed?

raldi, 2022-03-14:
I googled "windows python" and ended up here: https://www.python.org/downloads/release/python-3102/

The download link marked "Recommended" was the "Windows installer (64-bit)" at https://www.python.org/ftp/python/3.10.2/python-3.10.2-amd64.exe

After installing it, I have a Python section of the Start menu that, when clicked, opens an interpreter, but I notice if I instead just open a cmd shell and type "python" or "python3" it just returns the prompt with no output and fires up Python's page at the Microsoft Store.

raldi, 2022-03-14:
It appears to be installed at [User]/AppData/Local/Programs/Python/Python310/python.exe

raldi, 2022-03-14:
I tried adding it to my path, but apparently paths don't actually work in Windows?

https://i.imgur.com/c4vEF2q.png

raldi, 2022-03-14:
Okay, looks like I needed to turn these off: https://i.imgur.com/FQaFPCF.png

raldi, 2022-03-14:
Confirmed, I can now bring up the GUI. So I guess we should either update the instructions or just make MT work a little harder to find python.exe.

mkinney, 2022-03-14:
Just released v1.0.2 of WinApp - see https://github.com/meshtastic/Meshtastic-gui-installer/releases/tag/winapp1.0.2 

This version checks `py --version`, then checks `python3 --version`, if the first check fails. If it cannot find a valid `python3`, it will let the user know in a message box and will not continue.

Let me know if you're ok closing this issue.

raldi, 2022-03-14:
I think so. You might have to field some support requests from people who say, "I got an error message telling me to install Python 3, but I *do* have it installed", but if that happens, you can link them here for instructions on how to fix their path.

mkinney, 2022-03-17:
Closing this issue for now. If you want, feel free to re-open if you experience the same issue. (or create a new issue)

closed 2022-03-17 by mkinney