meshtastic/Meshtastic-gui-installer #150: review/fix translated values

mkinney, 2022-03-14:
Create PR for translated values in fields.json. https://github.com/meshtastic/Meshtastic-gui-installer/blob/master/meshtastic_flasher/fields.json

The initial values were created using deepl.com, however, a native speaker might translate the values to something more reasonable.

No coding necessary for this task. Just github PRs and your native language knowledge.

mkinney, 2022-03-14:
These languages have been reviewed by a native speaker:

- [X] Deutsch / German / "de"
- [ ] "Español / Spanish / "es"
- [ ] Français / French / "fr"
- [ ] Italiano / Italian / "it"
- [ ] Polski / Polish / "pl"
- [ ] Chinese / "zh"
- [ ] русский / Russian "ru"
- [ ] 日本語 / Japanese "ja"

laosb, 2022-03-16:
I might be able to review Chinese translation. Not an EE/Radio guy though, so I won't be able to offer much advice on some deep tech terms.

laosb, 2022-03-16:
One question: Does the JSON have to use unicode escape strings for non-ASCII? Since Chinese characters are non-ASCII, it's very hard to inspect the JSON file directly on GitHub web interface.

mkinney, 2022-03-16:
That's great! @laosb 

I wrote a simple script to get around that `bin/show_entries_for_lang.py zh > ~/Desktop/zh.txt`

Here's the file:
[zh.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8259845/zh.txt)

Can you make changes to that file directly, then add an updated file to a comment?

laosb, 2022-03-16:
> That's great! @laosb If there's anything technical or you don't understand, feel free to either skip that one or ask questions. 
> 
> 
> 
> I wrote a simple script to get around that `bin/show_entries_for_lang.py zh > zh.txt`
> 
> 
> 
> Here's the file:
> 
> [zh.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8259845/zh.txt)
> 
> 
> 
> Can you make changes to that file directly, then add an updated file to a comment?
> 
> 
> 
> Edit: We found that "canned messages" does not really translate well. Perhaps use "predetermined messages" when you see that text, since that is what the module is intended to do.

That's good, I will try to review this later today.

Daroes433, 2022-03-16:
Hi, I am a native Spanish, I can proofread the text

Daroes433, 2022-03-16:
How do you do this? (I'm new) I was making changes to later make a propose changes. It's okay like that?

mkinney, 2022-03-16:
> How do you do this? (I'm new) I was making changes to later make a propose changes. It's okay like that?

How about using this file to review and edit your changes?
[es.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8263455/es.txt)

When done, add the updated file to a comment? (just drag file into the comment section)

mkinney, 2022-03-16:
> Great!
> One question, there are signs in Spanish (for example ñ ¿ ´) that must be activated or notified of the use of utf-8 in order for them to appear. Will the text appear only in the installer GUI or also on the device screen? Thank you

These changes are for the Meshtastic-flasher program. You should be able to use your normal text editor, that has those specific signs. I think the Meshtastic device needs more work to handle Unicode on the displayed messages. However, the clients like Android and iOS might handle them ok, but I’m not 100% sure. You can test it yourself. 😀

laosb, 2022-03-16:
@mkinney

A bit more effort than I anticipated, but here we are:

[zh.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8264367/zh.txt)

Some problems to note:

- The `field:coding_rate key:description` has an example explaining the concept of denominator. However, in the original English text, it says `ie for 4/8, the value is 8. 5/8 the value is 5`, which is not correct. I believe the correct expression would be `ie for 4/8, the value is 8. 4/5 the value is 5`. Didn't touch the original text but the translation is corrected.
- The phrasing `field:ext_notification_module_active key:description` is misleading to me (and DeepL it seems). It seems to imply that the device generates a notification message when the set GPIO is active-high / active-low, where it's actually the very opposite. I would recommend change the wording a bit to clarify.

And it would be better to make this version of Chinese `zh-cn` since it's done according to Simplified Chinese in Mainland China. That will make space for other variants, like `zh-tw` which is a different dialect of Chinese used in Taiwan. I can help with `zh-tw` variant though it would be better if some native speakers of this dialect is interested in doing it.

mkinney, 2022-03-16:
@laosb Thank you for your efforts! I really appreciate it.

I've created a new issue to look into the coding_rate. Thanks for catching that.

I've also created an issue in deepl about the "Simplified" comment. (makes sense to me)

I'll take a look at the file you uploaded and see if I have any questions. Again, *THANKS!*. :-)

Daroes433, 2022-03-16:
Done, here the file with corrections

[es.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8266194/es.txt)

Daroes433, 2022-03-16:
Corrections are already made. Greetings
[es.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8266744/es.txt)

mkinney, 2022-03-17:
Le Andre to review the Estonian: 
[et.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8292482/et.txt)

Cavemantaco to review the Romanian:
[ro.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8292498/ro.txt)

mkinney, 2022-03-21:
The updated Russian (after review) by Dmitri:
[ru2.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8316749/ru2.txt)

mkinney, 2022-03-30:
Updated Estonian

mkinney, 2022-03-30:
Updated what work was done on Romanian

mkinney, 2022-03-31:
Here are updated files, if anyone wants to work on an individual language and they do not want to do pull requests:

[ja.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8392213/ja.txt)
[ro.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8392214/ro.txt)
[pl.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8392215/pl.txt)
[fr.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8392216/fr.txt)
[it.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8392219/it.txt)

wkawecki, 2022-04-13:
Sorry, my mistake - wrong file. This should be correct
[pl.txt](https://github.com/meshtastic/Meshtastic-gui-installer/files/8484488/pl.txt)

mkinney, 2022-04-14:
No worries. Thank you.