meshtastic/Meshtastic-gui-installer #152: Windows installer can't find its own firmware file

raldi, 2022-03-14:
I was able to get pretty far in the Windows flasher just now, but when it got to the last step, it failed with "No such file or directory" on the firmware binary: https://i.imgur.com/WYDpuDM.png

Here's what was going on over in the console: https://i.imgur.com/hZTgnUs.png

mkinney, 2022-03-14:
Tip: You can drag/drop an image into the GitHub issue, wait a minute and the image is hosted on GitHub.

You could try going into "Advanced Options" and clearing the Firmware files, then re-trying the "Get Versions". It is possible the downloaded zip file or extracted file got corrupted.

raldi, 2022-03-14:
Actually, just closing the flasher and relaunching it, the second time around it was able to find the files. Maybe it's an issue that only happens the first run?

mkinney, 2022-03-17:
Closing this issue for now. If you want, feel free to re-open if you experience the same issue. (or create a new issue)

closed 2022-03-17 by mkinney