meshtastic/Meshtastic-gui-installer #156: Error when opening Meshtastic Flasher mac OS

w3bsa, 2022-03-14:
I get the following error when trying to open meshtastic flasher on Mac OS Big Sur 11.6

Error Message:
The action “Run Shell Script” encountered an error: “Traceback (most recent call last):
File "/Users/*********/meshtastic-flasher/venv/bin/meshtastic-flasher", line 5, in <module>
from meshtastic_flasher.main import main
File "/Users/*******/meshtastic-flasher/venv/lib/python3.7/site-packages/meshtastic_flasher/main.py", line 7, in <module>
from PySide6.QtGui import QIcon
ImportError: could not import module 'PySide6.QtCore'”

mkinney, 2022-03-14:
I need to check if Pyside6 is supported on python 3.7 and/or BigSur.

mkinney, 2022-03-27:
I came across this little nugget:

```
Try to build it on as old a version of macOS as possible. This improves the compatibility of your app with older versions of macOS. For example, an app built on macOS 10.10 is most likely to also run on 10.14, but not the other way around. Most people use virtual machines to run old versions of macOS.
```