meshtastic/Meshtastic-gui-installer #173: tbeam flash issue

mdezendorf, 2022-03-20:
I picked up 2 tbeams [these ](https://usa.banggood.com/LILYGO-TTGO-T-Beam-ESP32-433-or-868-or-915-or-923Mhz-V1_1-WiFi-Wireless-bluetooth-Module-GPS-NEO-6M-SMA-LORA32-18650-Battery-Holder-With-OLED-p-1545070.html?cur_warehouse=USA&ID=510803) from Banggood and put the screens on but neither one will light up. I used the Meshtastic flasher using version 1.2.58.6af1822  with device tbeam and now a red light blinks but the screen still doesn't turn on. Any and all help would be appreciated, sorry BTW if I am posting out of line I am just trying to get help and the Meshtastic website said "Issues?
If you run into an issue, please create a ticket here: [Flasher Issues](https://github.com/meshtastic/Meshtastic-gui-installer/issues)"

caveman99, 2022-03-20:
If the red light under the screen (not the one next to the GPS module) powers on it should work. Maybe you got the screen on wrong. Can you post a photo of your screen assembly?

mdezendorf, 2022-03-20:
[image]

mdezendorf, 2022-03-20:
[image]
better view

caveman99, 2022-03-21:
You got the right pins and the right display type (2 common pitfalls) but the solder points look less than perfect. It's enough if one of them doeasn't have a 100% contact. Try to reflow them carefully (heat to about 200C and add a tiny bit of solder once the existing blob melts. Then wait till it forms a perfect cone from the eye to the wire)

mdezendorf, 2022-03-22:
O.O You dear @caveman99 are a lifesaver, sure as hell . . . . case closed, THANK YOU!!!!!!

closed 2022-03-22 by mdezendorf