meshtastic/Meshtastic-gui-installer #181: Bad value write when use a float value in adc_multiplier_override field

amerinoj, 2022-03-25:
Hello I was playing with POWER settings and realize that the "adc multiplier override" don't work with float values.

For example if you try to put a 1.2 as value in POWER(adc_multiplier_override) this send  " Set adc_multiplier_override to 0 " .
This field only work with integer values and should work with float values.

caveman99, 2022-03-25:
The protobufs define it as "float adc_multiplier_override" - so the value can be a float. Have you tried setting it with the meshtastic CLI?

amerinoj, 2022-03-25:
In meshtastic CLI working fine, but when you put a decimal in GUI send 0

mkinney, 2022-03-25:
Thanks for reporting the issue. Will be fixed in next release.

mkinney, 2022-03-25:
Fixed in latest release: https://github.com/meshtastic/Meshtastic-gui-installer/releases/tag/1.0.102

PR #182 (2022-03-25): allow floating point values
Fix #181
closed 2022-03-25 by mkinney