meshtastic/Meshtastic-gui-installer #193: Meshtastic Flasher v1.3alpha 13 is not detecting tbeam v1.1 M8 correctly in detect device

rwcts, 2022-04-09:
Running Meshtastic Flasher v1.3alpha 13
Windows 11 64bit
T-Beam v1.1 M8N w/OLED

I click get versions and then i select 1.3.5.e5b19fd
It I then click Detect Device it then pops up asking if the device currently has meshtastic 1.2 or newer and i select yes.

It then says::
[image]

Looks like a RAK 11200 was detected

Verify it is not in BOOT mode

There should be a single, solid red light.
There should not be any other lights, solid, nor flashing

If not, disconnect the device and provide a jumper between GND and BOOT0 pins while device is plugged in.

I then click OK and it shows the proper COM port but the device is showing tlora-v1. In a previous test it it chose RAK11200.

rwcts, 2022-04-09:
Debug widow output.
C:\Users\User>meshtastic-flasher
start of get_versions
403 {"message": "API rate limit exceeded for MYIPIShidden. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.)", "documentation_url": "https://docs.github.com/rest/overview/resources-in-the-rest-api#rate-limiting"}
got versions
start of detect
Detected T-Lora
Detected T-Lora
Detected T-Beam
Detected T-Beam
Detected T-Lora
Detected RAK 11200
Detected T-Beam
Detected T-Beam
User confirmed the device has Meshtastic so we will check using the Meshtastic python method
Warning: Could not find any ports using the Meshtastic python autodetection method.
from find_ports ports:['COM15', 'COM3', 'COM5', 'COM6']
Looks like a RAK 11200, ensure in boot mode (single red solid light, no green nor blue lights).
See https://docs.rakwireless.com/assets/images/wisblock/rak11200/quickstart/rak11200-Boot0-for-flashing.png
self.detected_list:[]
end of detect

garthvh, 2022-04-20:
What software is on the device?

rwcts, 2022-05-24:
> What software is on the device?

1.3.5.e5b19fd

garthvh, 2022-08-27:
Consolidating detection issue to https://github.com/meshtastic/Meshtastic-gui-installer/issues/233

closed 2022-08-27 by garthvh