meshtastic/Meshtastic-gui-installer #198: channel info is erased when updating Heltec Lora 32 v2.1

rwcts, 2022-04-17:
I did a --info before flashing and update only using the meshtastic flasher alpha 13 (i downloaded the newest version alpha 14 but still showing as 13?) Below is the --info output before and after UPDATE flash from 1.3.5.e5b19fd and 1.3.7.bb22b6e

Owner: NODE3 (E3)
My info: { "myNodeNum": 2130616524, "firmwareVersion": "1.3.5.e5b19fd", "rebootCount": 295, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 36.0, "airUtilTx": 1.0270278 }
Nodes in mesh:  {'num': 2130616524, 'user': {'id': '!7eff15fc', 'longName': 'NODE3', 'shortName': 'E3', 'macaddr': '94:b9:11:ff:1a:fc', 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'deviceMetrics': {'channelUtilization': 43.056667, 'airUtilTx': 0.4696111}}  {'num': 4145656180, 'user': {'id': '!f71e4554', 'longName': 'NODE3', 'shortName': 'E8', 'macaddr': '30:c6:f7:7e:aa:54', 'hwModel': 'TBEAM'}, 'snr': 11.0}  {'num': 4145918432, 'user': {'id': '!f71e4e20', 'longName': 'NODE10', 'shortName': 'E10', 'macaddr': '30:c6:f7:99:4e:90', 'hwModel': 'TBEAM'}, 'position': {'time': 1650009774}, 'snr': 3.25, 'lastHeard': 1650006999, 'deviceMetrics': {'batteryLevel': 100, 'voltage': 4.183, 'channelUtilization': 59.52, 'airUtilTx': 28.962029}}

Preferences: { "screenOnSecs": 30, "phoneTimeoutSecs": 900, "lsSecs": 300, "region": "US", "positionFlags": 35, "hopLimit": 3, "isPowerSaving": true }

Channels:
  PRIMARY psk=secret { "modemConfig": "LongSlow", "psk": "KEYHIDDEN", "name": "NAME", "id": 1234, "uplinkEnabled": true, "downlinkEnabled": true }
  SECONDARY psk=secret { "psk": "KEYHIDDEN", "name": "admin", "uplinkEnabled": true, "downlinkEnabled": true }

Primary channel URL: URLHIDDEN
Complete URL (includes all channels): URL HIDDEN

C:\Users\User\Downloads\Meshtastic-gui-installer-1.3alpha.14\Meshtastic-gui-installer-1.3alpha.14>meshtastic --info --port COM9
Connected to radio

Owner: NODE3 (E3)
My info: { "myNodeNum": 2130616524, "firmwareVersion": "1.3.7.bb22b6e", "rebootCount": 298, "bitrate": 17.08847, "messageTimeoutMsec": 300000, "minAppVersion": 20200, "maxChannels": 8, "hasWifi": true, "channelUtilization": 36.0, "airUtilTx": 1.0270278 }
Nodes in mesh:  {'num': 2130616524, 'user': {'id': '!7eff15fc', 'longName': 'NODE33', 'shortName': 'E3', 'macaddr': '94:b9:11:ff:1a:fc', 'hwModel': 'HELTEC_V2_1'}, 'position': {}, 'deviceMetrics': {'channelUtilization': 43.056667, 'airUtilTx': 0.4696111}}  {'num': 4145956580, 'user': {'id': '!f71e4554', 'longName': 'NODE8', 'shortName': 'E8', 'macaddr': '30:c6:f7:11:45:54', 'hwModel': 'TBEAM'}, 'snr': 11.0}  {'num': 4145951432, 'user': {'id': '!f71e4e20', 'longName': 'NODE10', 'shortName': 'E10', 'macaddr': '30:c6:f7:1e:aa:11', 'hwModel': 'TBEAM'}, 'position': {'time': 1650096774}, 'snr': 3.25, 'lastHeard': 1650006999, 'deviceMetrics': {'batteryLevel': 100, 'voltage': 4.183, 'channelUtilization': 59.52, 'airUtilTx': 28.962029}}

Preferences: {}

Channels:

Primary channel URL: https://www.meshtastic.org/d/#

----------------Below is meshtastic flasher output-------------------------
start of get_versions
tag:v1.3.7.bb22b6e populate in dropdown?:True
tag:v1.3.6.f511bab populate in dropdown?:True
tag:v1.3.5.e5b19fd populate in dropdown?:True
tag:v1.2.59.d81c1c0 populate in dropdown?:False
tag:v1.2.58.6af1822 populate in dropdown?:False
tag:v1.3.4.2b20bf3 populate in dropdown?:True
tag:v1.3.3.2fe124e populate in dropdown?:True
tag:v1.2.57.f7c6955 populate in dropdown?:False
tag:v1.2.56.596a73c populate in dropdown?:False
tag:v1.2.55.9db7c62 populate in dropdown?:False
tag:v1.2.54.288f2be populate in dropdown?:False
tag:v1.2.53.19c1f9f populate in dropdown?:False
tag:v1.2.52.b63802c populate in dropdown?:False
tag:v1.2.51.f9ff06b populate in dropdown?:False
tag:v1.2.50.41dcfdd populate in dropdown?:False
tag:v1.2.49.5354c49 populate in dropdown?:False
tag:v1.2.48.371335e populate in dropdown?:False
tag:v1.2.47 populate in dropdown?:False
tag:v1.2.46.dce2fe4 populate in dropdown?:False
tag:v1.2.45.b674054 populate in dropdown?:False
tag:v1.2.44.f2c9c55 populate in dropdown?:False
Need to download...
zip_file_url:https://github.com/meshtastic/Meshtastic-device/releases/download/v1.3.7.bb22b6e/firmware-1.3.7.bb22b6e.zip
downloading...
done downloading
Unzipping files now...
done unzipping
got versions
OK button was clicked in advanced options
start of detect
Detected Meshtastic DIY
Detected T-Lora
Detected Heltec
Detected Heltec
Detected Heltec
User confirmed the device has Meshtastic so we will check using the Meshtastic python method
Warning: Could not find any ports using the Meshtastic python autodetection method.
from find_ports ports:['COM3', 'COM5', 'COM6', 'COM9']
Getting version and hwModel from Meshtastic python library
Exception:Timed out waiting for connection completion
filenames:['1.3.7.bb22b6e\\firmware-heltec-v1-1.3.7.bb22b6e.bin', '1.3.7.bb22b6e\\firmware-heltec-v2.0-1.3.7.bb22b6e.bin', '1.3.7.bb22b6e\\firmware-heltec-v2.1-1.3.7.bb22b6e.bin', '1.3.7.bb22b6e\\firmware-meshtastic-diy-v1-1.3.7.bb22b6e.bin', '1.3.7.bb22b6e\\firmware-rak11200-1.3.7.bb22b6e.bin', '1.3.7.bb22b6e\\firmware-tbeam-1.3.7.bb22b6e.bin', '1.3.7.bb22b6e\\firmware-tbeam0.7-1.3.7.bb22b6e.bin', '1.3.7.bb22b6e\\firmware-tlora-v1-1.3.7.bb22b6e.bin', '1.3.7.bb22b6e\\firmware-tlora-v2-1-1.6-1.3.7.bb22b6e.bin', '1.3.7.bb22b6e\\firmware-tlora-v2-1.3.7.bb22b6e.bin', '1.3.7.bb22b6e\\firmware-tlora_v1_3-1.3.7.bb22b6e.bin']
self.detected_list:[]
end of detect
Flash was clicked.
update only is checked
User confirmed they want to flash
do esptool update_only:True
finished:True
OK button was clicked in esptool form
in settings
Warning: Multiple serial ports were detected so one serial port must be specified with the '--port'.
  Ports detected:['COM3', 'COM5', 'COM6', 'COM9']
User does not want to use a fake device
closing
closed Settings

rwcts, 2022-04-17:
The odd thing is that the screen still shows the proper channel name

rwcts, 2022-04-17:
[image]

garthvh, 2022-08-27:
Closing as 1.2 is not receiving any updates

closed 2022-08-27 by garthvh