meshtastic/Meshtastic-gui-installer #200: "Failed to write to target RAM"  on a MacOS and T-Beam

alexanderthefox, 2022-04-20:
Hello Meshtastic community!
Today I installed the Meshtastic Flasher (version v1.0.103) on my MacBook Pro (MacOs Mojave - 10.1.4.6). 
When trying to flash Mashtastic on my T-Beam (which I just got delivered and there is no Meshtastic preinstalled) I got the error message "Failed to write to target RAM (result was 01070000: Operation timed out)". 
I tried several times to flash by disconnecting/re-connceting the USB cable from the T-Beam, unfortunately without success...;-(
Can someone here help me to get Meshtastic on the T-Beam?
Many thanks in advance!

Best regards
Alexander
---
Here is the protocol output from the installer in full length:
Serial port /dev/cu.usbmodem51850276131
Connecting...
.
Detecting chip type...
Unsupported detection protocol, switching and trying again...
Connecting...
.
.
.
Detecting chip type...
ESP32

Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 30:c6:f7:1e:ad:74
Uploading stub...
Traceback (most recent call last):
File "/Users/alexander/meshtastic/venv/lib/python3.10/site-packages/meshtastic_flasher/esptool_form.py", line 70, in run

esptool.main(command)
File "/Users/alexander/meshtastic/venv/bin/esptool.py", line 4710, in main

esp = esp.run_stub()
File "/Users/alexander/meshtastic/venv/bin/esptool.py", line 891, in run_stub

self.mem_block(stub[field][from_offs:to_offs], seq)
File "/Users/alexander/meshtastic/venv/bin/esptool.py", line 755, in mem_block

return self.check_command("write to target RAM", self.ESP_MEM_DATA,
File "/Users/alexander/meshtastic/venv/bin/esptool.py", line 505, in check_command

raise FatalError.WithResult('Failed to %s' % op_description, status_bytes)
esptool
.
FatalError
:
Failed to write to target RAM (result was 01070000: Operation timed out)

garthvh, 2022-04-20:
Do you have an earlier version of Meshtastic on the device or SoftRF? Have you installed a usb to serial driver?

garthvh, 2022-04-20:
There are instructions and links to the two drivers here https://meshtastic.org/docs/getting-started/flashing-firmware it is likely you need the 9102 driver, and you can not have both installed at the same time.

The usb to serial chip model with drivers included in the OS has been unavailable for about a year due to the global chip shortage so new devices come with a 9102 chip most of the time now.

You also may have to use the install script as we have not nailed the upgrade from a non Meshtastic firmware experience in the flasher yet.

alexanderthefox, 2022-04-20:
Hi garthvh, thanks for that hint. 
Pls see below some additional questions.

> There are instructions and links to the two drivers here https://meshtastic.org/docs/getting-started/flashing-firmware it is likely you need the 9102 driver, and you can not have both installed at the same time.

_Should I download that driver and is it necessary to de-install the older version? If so, how should I do that?_
> The usb to serial chip model with drivers included in the OS has been unavailable for about a year due to the global chip shortage so new devices come with a 9102 chip most of the time now.
>
> You [...]
_What script do you mean?_

alexanderthefox, 2022-04-20:
Hi garthvh,

I installed the 9102 driver. But this didn't solved the problem. 
The report of meshtastic flasher gave me that:

Step 1/4 esp32 full
ESPTOOL Using command:--port /dev/cu.usbmodem51850276131 erase_flash
esptool.py v3.3
Serial port /dev/cu.usbmodem51850276131
Traceback (most recent call last):
File "/Users/alexander/meshtastic/venv/lib/python3.10/site-packages/serial/serialposix.py", line 322, in open

self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
OSError
:
[Errno 16] Resource busy: '/dev/cu.usbmodem51850276131'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/Users/alexander/meshtastic/venv/lib/python3.10/site-packages/meshtastic_flasher/esptool_form.py", line 70, in run

esptool.main(command)
File "/Users/alexander/meshtastic/venv/bin/esptool.py", line 4687, in main

esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
File "/Users/alexander/meshtastic/venv/bin/esptool.py", line 110, in get_default_connected_device

_esp = ESPLoader.detect_chip(each_port, initial_baud, before, trace,
File "/Users/alexander/meshtastic/venv/bin/esptool.py", line 363, in detect_chip

detect_port = ESPLoader(port, baud, trace_enabled=trace_enabled)
File "/Users/alexander/meshtastic/venv/bin/esptool.py", line 320, in __init__

self._port = serial.serial_for_url(port)
File "/Users/alexander/meshtastic/venv/lib/python3.10/site-packages/serial/__init__.py", line 90, in serial_for_url

instance.open()
File "/Users/alexander/meshtastic/venv/lib/python3.10/site-packages/serial/serialposix.py", line 325, in open

raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil
.
SerialException
:
[Errno 16] could not open port /dev/cu.usbmodem51850276131: [Errno 16] Resource busy: '/dev/cu.usbmodem51850276131'

mkinney, 2022-04-20:
`Resource busy` might mean that there is some other software running in addition to m-flasher. Do you have anything like Cura (or any other app that might be using the serial port) running?

alexanderthefox, 2022-04-20:
> `Resource busy` might mean that there is some other software running in addition to m-flasher. Do you have anything like Cura (or any other app that might be using the serial port) running?

Hi mkinney,

right now it is the meshtastic-flasher is the only software, that is running and using the port. But when I'm looking at the system report, under USB2.0 Hub there are twi USB Single Serials with the same Vendor ID are listed.

garthvh, 2022-04-20:
Dupe of #159

garthvh, 2022-04-20:
The flasher uses the installed driver.

closed 2022-04-20 by garthvh