meshtastic/Meshtastic-gui-installer #208: Flasher errors on ttgo t-beams

SSQ00-MAD, 2022-04-30:
Hello

Could you please advice me on the ways how I can overcome an issue I've been running into.

I've already had 2 meshtastic t-beams that are working flawlessly. Recently I've decided to purchase another two. They came in with Meshtastic version 1.0.0 firmware on it, so I decided to upgrade them to  version 1.2.52. Get device does not recognise them, and after selecting t-beam from the list while flashing with Meshtastic Flasher v.1.0.105 on my Windows PC I run into the next message:

"Had a problem talking with the device. Maybe disconnect and reconnect the device? If the device is an nrf52 (T-Echo or RAK), then put in boot mode by pressing RST button twice."

Mine are t-beam devices and suggested click combo does not help. When attempting to flash after clicking "ok" in this message the Flasher returns following message on step 1 out of 4: "could not open port 'COM7': PermissionError(13, 'Access is denied.', None, 5)".

After several attempts with different cables and port options I still cannot flash these 2 t-beams successfully, and the devices themselves are not starting up with meshtastic logo and the previosly existing firmware 1.0.0. So I'm kinda stuck with two non working devices and I'm out of the ideas how to overcome it.

Could you please advice any steps I can take to make them function normally? I have a working t-beam with firmware 1.2.52 on it, so is there a way to make an image out of it & flash it into another t-beam?

Thanks!

lycaon42, 2022-05-24:
I'm having a similar issue.
When trying to update my firmware to version 1.2.25 using the meshtastic flasher, I got the same error:
 ""Had a problem talking with the device. Maybe disconnect and reconnect the device? If the device is an nrf52 (T-Echo or RAK), then put in boot mode by pressing RST button twice."

After clicking "ok" and attempting to flash, it gave the error:

"raise FatalError.WithResult('Failed to %s' % op_description, status_bytes)
esptool
.
FatalError
:
Failed to write to target RAM (result was 01070000: Operation timed out)"

I'm using a mac book air and have to use an adapter from usb-c to usb 2.0 to micro usb in order to connect to the meshtastic t-beam node. 
[image]
And I was worried this was part of this issue.
Any help would be appreciated if you've already found a solution @SSQ00-MAD
Or any kind soul that stumbles across this and knows the solution!

SSQ00-MAD, 2022-05-30:
> I'm having a similar issue. When trying to update my firmware to version 1.2.25 using the meshtastic flasher, I got the same error: ""Had a problem talking with the device. Maybe disconnect and [...]

I've eventually managed to flush 1.2.52 onto my T-beam devices using ESPHome-Flasher-1.4.0 and 1.2.52 firmware available at https://meshtastic.org/downloads. 

So I think you can also give this flasher a try. Thanks

garthvh, 2022-08-27:
These look like USB to serial driver issues, try installing the CH340 driver

closed 2022-08-27 by garthvh