meshtastic/Meshtastic-gui-installer #210: bug: detecting t-lora v2.1 1.6.1 as tbeam and nano-g1

dcherrera, 2022-05-02:
when running the software in a background bash it detects my t-lore as a t-beam when auto detecting. when opening the software manually in terminal to get an output it detects as a nano-g1. I have tried 2 different cables and will try a couple more in the morning. now difference between cables when auto detecting with that bash script in the background.

pypi_version:1.0.105
running: 1.0.105
start of detect
group_to_search:dialout
Detected Nano G1
Detected T-Lora
Detected T-Beam
Detected T-Beam
Detected T-Beam
Detected T-Lora
Detected T-Lora
Detected T-Beam
Detected T-Lora
User confirmed the device has Meshtastic so we will check using the Meshtastic python method
Warning: Could not find any ports using the Meshtastic python autodetection method.
from find_ports ports:['/dev/ttyACM0']
Getting version and hwModel from Meshtastic python library
filenames:['1.2.63.9879494/firmware-tlora-v2-1.2.63.9879494.bin', '1.2.63.9879494/firmware-tbeam0.7-1.2.63.9879494.bin', '1.2.63.9879494/firmware-tlora-v1-1.2.63.9879494.bin', '1.2.63.9879494/firmware-heltec-v2.1-1.2.63.9879494.bin', '1.2.63.9879494/firmware-tbeam-1.2.63.9879494.bin', '1.2.63.9879494/firmware-tlora-v2-1-1.6-1.2.63.9879494.bin', '1.2.63.9879494/firmware-rak11200-1.2.63.9879494.bin', '1.2.63.9879494/firmware-tlora_v1_3-1.2.63.9879494.bin', '1.2.63.9879494/firmware-heltec-v2.0-1.2.63.9879494.bin', '1.2.63.9879494/firmware-meshtastic-diy-v1-1.2.63.9879494.bin', '1.2.63.9879494/firmware-heltec-v1-1.2.63.9879494.bin', '1.2.63.9879494/firmware-nano-g1-1.2.63.9879494.bin']
self.detected_list:[]
end of detect

JebBaker, 2022-08-11:
I have a similar issue.  My t-beam gets detected as tlora-v1 same detect out put.

garthvh, 2022-08-11:
Do your devices already have a version of meshtastic installed @dcherrera & @JebBaker?

The detection does not work great if you don't have meshtastic as they all identify themselves essentially as the same dev board.

JebBaker, 2022-08-11:
That did it.  As soon as I flashed it, which I assumed the board was flashed already, then the flasher allowed me to go into the device settings.  Thanks.

garthvh, 2022-08-27:
Consolidating detection issue to https://github.com/meshtastic/Meshtastic-gui-installer/issues/233

closed 2022-08-27 by garthvh