meshtastic/Meshtastic-gui-installer #214: module 'esptool' has no attribute 'main'

j4nr6n, 2022-05-16:
I'm running Ubuntu 20.04 and used [these installation instructions](https://meshtastic.org/docs/getting-started/flashing-firmware/meshtastic-flasher#install-app). Not sure what changed. I just flashed two other tbeams about a week ago.

```
Step 1/4 esp32 full
ESPTOOL Using command:--port /dev/ttyACM0 erase_flash
Traceback (most recent call last):
File "/home/me/meshtastic/venv/lib/python3.8/site-packages/meshtastic_flasher/esptool_form.py", line 70, in run

esptool.main(command)
AttributeError
:
module 'esptool' has no attribute 'main'
```

j4nr6n, 2022-05-16:
Seems related to [the new version of esptools](https://github.com/espressif/esptool/releases/tag/v4.0) I was able to get things running again by running `pip install -Iv esptool==3.3.1` to downgrade esptools to the previous version.

mkinney, 2022-05-19:
@thebentern Had a nice temp fix for this. Will create a new version as soon as github stops giving me 500 errors.

mkinney, 2022-05-23:
GitHub Actions is now working again. Just released a new version. Can you try updating?

garthvh, 2022-08-27:
I also had to install a specific version of esp tool to get this to work on my MacBook Air