meshtastic/Meshtastic-gui-installer #219: ERROR: Cannot install meshtastic-flasher==1.0.100

AusUser5, 2022-06-14:
(venv) C:\Users\test\meshtastic>python --version
Python 3.10.5

` Downloading meshtastic_flasher-1.0.32-py3-none-any.whl (20 kB)
  Downloading meshtastic_flasher-1.0.31-py3-none-any.whl (20 kB)
ERROR: Cannot install meshtastic-flasher==1.0.100, meshtastic-flasher==1.0.101, meshtastic-flasher==1.0.102, meshtastic-flasher==1.0.103, meshtastic-flasher==1.0.104, meshtastic-flasher==1.0.105, meshtastic-flasher==1.0.106, meshtastic-flasher==1.0.31, meshtastic-flasher==1.0.32, meshtastic-flasher==1.0.33, meshtastic-flasher==1.0.34, meshtastic-flasher==1.0.36, meshtastic-flasher==1.0.37, meshtastic-flasher==1.0.38, meshtastic-flasher==1.0.39, meshtastic-flasher==1.0.40, meshtastic-flasher==1.0.41, meshtastic-flasher==1.0.43, meshtastic-flasher==1.0.44, meshtastic-flasher==1.0.45, meshtastic-flasher==1.0.46, meshtastic-flasher==1.0.47, meshtastic-flasher==1.0.48, meshtastic-flasher==1.0.49, meshtastic-flasher==1.0.50, meshtastic-flasher==1.0.51, meshtastic-flasher==1.0.52, meshtastic-flasher==1.0.53, meshtastic-flasher==1.0.54, meshtastic-flasher==1.0.55, meshtastic-flasher==1.0.56, meshtastic-flasher==1.0.57, meshtastic-flasher==1.0.58, meshtastic-flasher==1.0.59, meshtastic-flasher==1.0.60, meshtastic-flasher==1.0.61, meshtastic-flasher==1.0.62, meshtastic-flasher==1.0.63, meshtastic-flasher==1.0.64, meshtastic-flasher==1.0.65, meshtastic-flasher==1.0.66, meshtastic-flasher==1.0.67, meshtastic-flasher==1.0.68, meshtastic-flasher==1.0.69, meshtastic-flasher==1.0.70, meshtastic-flasher==1.0.71, meshtastic-flasher==1.0.72, meshtastic-flasher==1.0.73, meshtastic-flasher==1.0.74, meshtastic-flasher==1.0.75, meshtastic-flasher==1.0.76, meshtastic-flasher==1.0.77, meshtastic-flasher==1.0.78, meshtastic-flasher==1.0.79, meshtastic-flasher==1.0.80, meshtastic-flasher==1.0.81, meshtastic-flasher==1.0.82, meshtastic-flasher==1.0.83, meshtastic-flasher==1.0.84, meshtastic-flasher==1.0.85, meshtastic-flasher==1.0.87, meshtastic-flasher==1.0.88, meshtastic-flasher==1.0.89, meshtastic-flasher==1.0.90, meshtastic-flasher==1.0.91, meshtastic-flasher==1.0.92, meshtastic-flasher==1.0.93, meshtastic-flasher==1.0.94, meshtastic-flasher==1.0.95, meshtastic-flasher==1.0.96, meshtastic-flasher==1.0.97, meshtastic-flasher==1.0.98 and meshtastic-flasher==1.0.99 because these package versions have conflicting dependencies.

The conflict is caused by:
    meshtastic-flasher 1.0.106 depends on pyside6
    meshtastic-flasher 1.0.105 depends on pyside6
    meshtastic-flasher 1.0.104 depends on pyside6
    meshtastic-flasher 1.0.103 depends on pyside6
    meshtastic-flasher 1.0.102 depends on pyside6
    meshtastic-flasher 1.0.101 depends on pyside6
    meshtastic-flasher 1.0.100 depends on pyside6
    meshtastic-flasher 1.0.99 depends on pyside6
    meshtastic-flasher 1.0.98 depends on pyside6
    meshtastic-flasher 1.0.97 depends on pyside6
    meshtastic-flasher 1.0.96 depends on pyside6
    meshtastic-flasher 1.0.95 depends on pyside6
    meshtastic-flasher 1.0.94 depends on pyside6
    meshtastic-flasher 1.0.93 depends on pyside6
    meshtastic-flasher 1.0.92 depends on pyside6
    meshtastic-flasher 1.0.91 depends on pyside6
    meshtastic-flasher 1.0.90 depends on pyside6
    meshtastic-flasher 1.0.89 depends on pyside6
    meshtastic-flasher 1.0.88 depends on pyside6
    meshtastic-flasher 1.0.87 depends on pyside6
    meshtastic-flasher 1.0.85 depends on pyside6
    meshtastic-flasher 1.0.84 depends on pyside6
    meshtastic-flasher 1.0.83 depends on pyside6
    meshtastic-flasher 1.0.82 depends on pyside6
    meshtastic-flasher 1.0.81 depends on pyside6
    meshtastic-flasher 1.0.80 depends on pyside6
    meshtastic-flasher 1.0.79 depends on pyside6
    meshtastic-flasher 1.0.78 depends on pyside6
    meshtastic-flasher 1.0.77 depends on pyside6
    meshtastic-flasher 1.0.76 depends on pyside6
    meshtastic-flasher 1.0.75 depends on pyside6
    meshtastic-flasher 1.0.74 depends on pyside6
    meshtastic-flasher 1.0.73 depends on pyside6
    meshtastic-flasher 1.0.72 depends on pyside6
    meshtastic-flasher 1.0.71 depends on pyside6
    meshtastic-flasher 1.0.70 depends on pyside6
    meshtastic-flasher 1.0.69 depends on pyside6
    meshtastic-flasher 1.0.68 depends on pyside6
    meshtastic-flasher 1.0.67 depends on pyside6
    meshtastic-flasher 1.0.66 depends on pyside6
    meshtastic-flasher 1.0.65 depends on pyside6
    meshtastic-flasher 1.0.64 depends on pyside6
    meshtastic-flasher 1.0.63 depends on pyside6
    meshtastic-flasher 1.0.62 depends on pyside6
    meshtastic-flasher 1.0.61 depends on pyside6
    meshtastic-flasher 1.0.60 depends on pyside6
    meshtastic-flasher 1.0.59 depends on pyside6
    meshtastic-flasher 1.0.58 depends on pyside6
    meshtastic-flasher 1.0.57 depends on pyside6
    meshtastic-flasher 1.0.56 depends on pyside6
    meshtastic-flasher 1.0.55 depends on pyside6
    meshtastic-flasher 1.0.54 depends on pyside6
    meshtastic-flasher 1.0.53 depends on pyside6
    meshtastic-flasher 1.0.52 depends on pyside6
    meshtastic-flasher 1.0.51 depends on pyside6
    meshtastic-flasher 1.0.50 depends on pyside6
    meshtastic-flasher 1.0.49 depends on pyside6
    meshtastic-flasher 1.0.48 depends on pyside6
    meshtastic-flasher 1.0.47 depends on pyside6
    meshtastic-flasher 1.0.46 depends on pyside6
    meshtastic-flasher 1.0.45 depends on pyside6
    meshtastic-flasher 1.0.44 depends on pyside6
    meshtastic-flasher 1.0.43 depends on pyside6
    meshtastic-flasher 1.0.41 depends on pyside6
    meshtastic-flasher 1.0.40 depends on pyside6
    meshtastic-flasher 1.0.39 depends on pyside6
    meshtastic-flasher 1.0.38 depends on pyside6
    meshtastic-flasher 1.0.37 depends on pyside6
    meshtastic-flasher 1.0.36 depends on pyside6
    meshtastic-flasher 1.0.34 depends on pyside6
    meshtastic-flasher 1.0.33 depends on pyside6
    meshtastic-flasher 1.0.32 depends on pyside6
    meshtastic-flasher 1.0.31 depends on pyside6
`
Tried pip install meshtastic-flasher --no-cache-dir same issue

mkinney, 2022-06-14:
Can you run this:

`pip install pyside6`

AusUser5, 2022-06-14:
`(venv) C:\Users\test\meshtastic>pip install pyside6
ERROR: Could not find a version that satisfies the requirement pyside6 (from versions: none)
ERROR: No matching distribution found for pyside6
WARNING: You are using pip version 22.0.4; `

mkinney, 2022-06-15:
That's not good. Hmm...

I wonder if it is case-sensitive. Can you try:

`pip install PySide6`

If that does not work, try:

`python -m pip install --upgrade pip`
`pip install --upgrade setuptools`
`pip install pyside6`

AusUser5, 2022-06-15:
`
(venv) C:\Users\test\meshtastic>pip install PySide6
ERROR: Could not find a version that satisfies the requirement PySide6 (from versions: none)
ERROR: No matching distribution found for PySide6`

python -m pip install --upgrade pip

`(venv) C:\Users\test\meshtastic>python -m pip install --upgrade pip
Requirement already satisfied: pip in c:\users\test\meshtastic\venv\lib\site-packages (22.1.2)`

pip install --upgrade setuptools

`(venv) C:\Users\test\meshtastic>pip install --upgrade  setuptools
Requirement already satisfied: setuptools in c:\users\test\meshtastic\venv\lib\site-packages (58.1.0)
Collecting setuptools
  Downloading setuptools-62.4.0-py3-none-any.whl (1.2 MB)
     ---------------------------------------- 1.2/1.2 MB 5.2 MB/s eta 0:00:00
Installing collected packages: setuptools
  Attempting uninstall: setuptools
    Found existing installation: setuptools 58.1.0
    Uninstalling setuptools-58.1.0:
      Successfully uninstalled setuptools-58.1.0
Successfully installed setuptools-62.4.0`

pip install pyside6

`(venv) C:\Users\test\meshtastic>pip install PySide6
ERROR: Could not find a version that satisfies the requirement PySide6 (from versions: none)
ERROR: No matching distribution found for PySide6`

AusUser5, 2022-06-16:
Trying a different system. I repeated the process on Raspberry PI using Python version 3.9.  Exact same errrors

mkinney, 2022-06-17:
Unfortunately, PI is not a supported platform (yet) for PySide6.

Can you try going back to Windows and using Python 3.9?

AusUser5, 2022-06-17:
Python 3.9.13 x86 same issue 
Python 3.8.6 x64 - worked

I read most of the packages have been compiled for Python 3.8 but not many for Python 3.9.

closed 2022-06-17 by AusUser5