meshtastic/Meshtastic-gui-installer #223: [1.0.106] Fixed GPS coordinates not set on new device

itmentor, 2022-07-01:
Environment : macOS 12.4
Installed version : 1.0.106
Device : Heltec Lora V2

After installing the firmware on the new terminal, follow the steps below

1. Set Fixed GPS Position in GUI and click Save button.
2. I tried to check the location saved with meshtastic --info, but the location is not saved and only the fixed position field is set in json
3. Manually set GPS coordinates with meshtastic --setlat --setlon
4. Check that the fields are properly filled with meshtastic --info.
5. Set the fixed gps position again in the GUI
4. Check that the fields have been changed normally with meshtastic --info.

If there is no lat, lon field related to the initial Fixed gps position in the GUI, it seems that the value cannot be written.

garthvh, 2022-07-01:
What firmware version is on the device

0xcodedead, 2022-07-01:
1.2.65

0xcodedead, 2022-07-01:
> What firmware version is on the device

1.2.65

garthvh, 2022-08-27:
1.2 is not receiving any updates, closing

closed 2022-08-27 by garthvh