meshtastic/Meshtastic-gui-installer #225: Not loading: pip install meshtastic-flasher

pcopa, 2022-07-03:
Sorry. I'm new to this and would appreciate any help in setting up meshtastic-flasher on my Win 10 64b.
This is the error I'm getting:

PS C:\pyvirt> python -m venv venv
PS C:\pyvirt> venv\Scripts\activate
(venv) PS C:\pyvirt> pip install meshtastic-flasher
Collecting meshtastic-flasher
  Using cached meshtastic_flasher-1.0.106-py3-none-any.whl (8.9 MB)
Collecting adafruit-nrfutil
  Using cached adafruit-nrfutil-0.5.3.post16.tar.gz (49 kB)
  Preparing metadata (setup.py) ... done
Collecting qt-material
  Using cached qt_material-2.12-py3-none-any.whl (1.7 MB)
Requirement already satisfied: psutil in c:\pyvirt\venv\lib\site-packages (from meshtastic-flasher) (5.9.1)
Collecting geocoder
  Using cached geocoder-1.38.1-py2.py3-none-any.whl (98 kB)
Collecting meshtastic>=1.2.95
  Using cached meshtastic-1.2.95-py3-none-any.whl (71 kB)
Requirement already satisfied: pyserial in c:\pyvirt\venv\lib\site-packages (from meshtastic-flasher) (3.5)
Collecting esptool==3.3.1
  Using cached esptool-3.3.1.tar.gz (213 kB)
  Preparing metadata (setup.py) ... done
Collecting pyside6
  Using cached PySide6-6.3.1-cp36-abi3-win_amd64.whl (66 kB)
Collecting PyGithub
  Using cached PyGithub-1.55-py3-none-any.whl (291 kB)
Requirement already satisfied: bitstring>=3.1.6 in c:\pyvirt\venv\lib\site-packages (from esptool==3.3.1->meshtastic-flasher) (3.1.9)
Collecting cryptography>=2.1.4
  Using cached cryptography-37.0.2-cp36-abi3-win_amd64.whl (2.4 MB)
Collecting ecdsa>=0.16.0
  Using cached ecdsa-0.17.0-py2.py3-none-any.whl (119 kB)
Requirement already satisfied: reedsolo<=1.5.4,>=1.5.3 in c:\pyvirt\venv\lib\site-packages (from esptool==3.3.1->meshtastic-flasher) (1.5.4)
Requirement already satisfied: pyyaml in c:\pyvirt\venv\lib\site-packages (from meshtastic>=1.2.95->meshtastic-flasher) (6.0)
Requirement already satisfied: tabulate>=0.8.9 in c:\pyvirt\venv\lib\site-packages (from meshtastic>=1.2.95->meshtastic-flasher) (0.8.10)
Requirement already satisfied: pyqrcode>=1.2.1 in c:\pyvirt\venv\lib\site-packages (from meshtastic>=1.2.95->meshtastic-flasher) (1.2.1)
Requirement already satisfied: pypubsub>=4.0.3 in c:\pyvirt\venv\lib\site-packages (from meshtastic>=1.2.95->meshtastic-flasher) (4.0.3)
Requirement already satisfied: dotmap>=1.3.14 in c:\pyvirt\venv\lib\site-packages (from meshtastic>=1.2.95->meshtastic-flasher) (1.3.30)
Requirement already satisfied: timeago>=1.0.15 in c:\pyvirt\venv\lib\site-packages (from meshtastic>=1.2.95->meshtastic-flasher) (1.0.15)
Requirement already satisfied: pexpect>=4.6.0 in c:\pyvirt\venv\lib\site-packages (from meshtastic>=1.2.95->meshtastic-flasher) (4.8.0)
Requirement already satisfied: protobuf>=3.13.0 in c:\pyvirt\venv\lib\site-packages (from meshtastic>=1.2.95->meshtastic-flasher) (4.21.2)
Collecting click>=5.1
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Collecting future
  Using cached future-0.18.2.tar.gz (829 kB)
  Preparing metadata (setup.py) ... done
Requirement already satisfied: six in c:\pyvirt\venv\lib\site-packages (from geocoder->meshtastic-flasher) (1.16.0)
Collecting requests
  Using cached requests-2.28.1-py3-none-any.whl (62 kB)
Collecting ratelim
  Using cached ratelim-0.1.6-py2.py3-none-any.whl (4.0 kB)
Collecting pynacl>=1.4.0
  Using cached PyNaCl-1.5.0-cp36-abi3-win_amd64.whl (212 kB)
Requirement already satisfied: pyjwt>=2.0 in c:\pyvirt\venv\lib\site-packages (from PyGithub->meshtastic-flasher) (2.4.0)
Collecting deprecated
  Using cached Deprecated-1.2.13-py2.py3-none-any.whl (9.6 kB)
Requirement already satisfied: shiboken6==6.3.1 in c:\pyvirt\venv\lib\site-packages (from pyside6->meshtastic-flasher) (6.3.1)
Collecting PySide6-Addons==6.3.1
  Using cached PySide6_Addons-6.3.1-cp36-abi3-win_amd64.whl (93.6 MB)
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    PySide6-Addons==6.3.1 from https://files.pythonhosted.org/packages/da/d2/eacfcd9a61dc0e04ce83197bfdff9be45be1882d0625df7d2bf1c13c014f/PySide6_Addons-6.3.1-cp36-abi3-win_amd64.whl#sha256=9a933fac75c85bea3a9e31b24926e06b1afebae23cd164f10f2001999ea2692c (from pyside6->meshtastic-flasher):
        Expected sha256 9a933fac75c85bea3a9e31b24926e06b1afebae23cd164f10f2001999ea2692c
             Got        e896c5eb628e95978f429a26820f7d7dba819db4cccfc882a04dddb81f598c5c

(venv) PS C:\pyvirt>`

Hesse68, 2022-07-03:
And you checked Python version before? (3.10.5)
python3 --version

pcopa, 2022-07-04:
yes its 3.10.5

pcopa, 2022-07-10:
Tried several options but no luck. Cant find "please update the hashes".
have tried the windows "meshtastic.exe" , but it times-out with error:

`PS D:\downloads> meshtastic.exe --info
Traceback (most recent call last):
  File "C:\Users\Schelto\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Schelto\AppData\Local\Programs\Python\Python310\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\Schelto\AppData\Local\Programs\Python\Python310\Scripts\meshtastic.exe\__main__.py", line 7, in <module>
  File "C:\Users\Schelto\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\__main__.py", line 929, in main
    common()
  File "C:\Users\Schelto\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\__main__.py", line 712, in common
    client = meshtastic.serial_interface.SerialInterface(args.port, debugOut=logfile, noProto=args.noproto)
  File "C:\Users\Schelto\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\serial_interface.py", line 57, in __init__
    StreamInterface.__init__(self, debugOut=debugOut, noProto=noProto, connectNow=connectNow)
  File "C:\Users\Schelto\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\stream_interface.py", line 50, in __init__
    self.connect()
  File "C:\Users\Schelto\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\stream_interface.py", line 74, in connect
    self._waitConnected()
  File "C:\Users\Schelto\AppData\Local\Programs\Python\Python310\lib\site-packages\meshtastic\mesh_interface.py", line 402, in _waitConnected
    raise Exception("Timed out waiting for connection completion")
Exception: Timed out waiting for connection completion
PS D:\downloads>`