meshtastic/Meshtastic-gui-installer #234: Multi Monitor issues under Windows 10

Noki, 2022-08-28:
The Flasher acts strange on multi monitor / multi graphic card issues under Windows 10. If the flasher is on a secondary display various dialogs (e.g. device settings with no connected device), detect device mouseover window, version dropdown content (only for < 1s) show up on the primary display. The info dialog however shows up on the same display as the flasher, so this might just be a programming issue.

Users in the chat pointed out that this might be related to having two displays connected to different graphic cards.

Expected behaviour would be that dialogs and mouseover notices always open on the same display as the main application window.

garthvh, 2022-08-28:
Is one of the gpu's integrated on you laptop or motherboard or are they both dedicated graphics cards?

Noki, 2022-08-28:
One is a dedicated Card (AMD), the other one is the Intel GPU integrated into the CPU. - It is a desktop system.

garthvh, 2022-10-10:
Know issue with two graphics cards

closed 2022-10-10 by garthvh