meshtastic/Meshtastic-gui-installer #237: no such file or directory exists

manu-mei-singh, 2022-09-26:
Using the directions on https://meshtastic.org/docs/getting-started/flashing-firmware/meshtastic-flasher  ...
Python version  3.7.3
- installed and ran metastatic installer
- Got latest version
- Found device (rak4631_5005) 
- From Terminal:
- User picked radio rak4631_5005
- - self.device_from_picker:rak4631_5005
- - index:1
- - end of detect
- - Flash was clicked.
- - update only is not checked
- -User confirmed they want to flash
- -Flash nrf52
- -copying file:1.3.41.80ddb81/firmware-rak4631_5005-1.3.41.80ddb81.uf2 to dest:/Volumes/RAK4631/firmware-rak4631_5005-1.3.41.80ddb81.uf2
- - Traceback (most recent call last):
- -  File "/Users/manusachdeva/Sites/localhost/mmsnode/virpy/venv/lib/python3.7/site-packages/meshtastic_flasher/form.py", line 1161, in flash_stuff
 - -self.flash_nrf52()
 - - File "/Users/manusachdeva/Sites/localhost/mmsnode/virpy/venv/lib/python3.7/site-packages/meshtastic_flasher/form.py", line 1067, in flash_nrf52
- -shutil.copyfile(uf2_file, dest)
- -File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/shutil.py", line 120, in copyfile
    with open(src, 'rb') as fsrc:
- -FileNotFoundError: [Errno 2] No such file or directory: '1.3.41.80ddb81/firmware-rak4631_5005-1.3.41.80ddb81.uf2'

Ok, the reason "fileNotFoundError" happened because the file in the drive was named "Current.uf2"  --> not sure if this specific to the set up on my RAK4631_5005 or is generalized to all RAK4631_5005

RESOLVED: 
 - Closed Meshtastic Flasher
 - Manually added firmware-rak4631_5005-1.3.41.80ddb81.uf2  to the RAK4631_5005 Folder/drive on MAC OSX (12.5)

garthvh, 2022-09-26:
You can't install 1.2 firmware with the latest flasher, there is a link to the pip command to install the last 1.2 flasher release here https://meshtastic.org/docs/1.2-End-of-life/

garthvh, 2022-09-26:
This is the 1.2 RAK firmware file firmware-rak4631-1.3.42.8ef5100.uf2

closed 2022-09-26 by garthvh

manu-mei-singh, 2022-09-26:
Ah I see. Great!  Yeah; I did the drag and drop Firmware update afterward the GUI.  Also, thanks for the quick response. Much appreciated. : )