meshtastic/Meshtastic-gui-installer #238: Windows 10 - Version won't populate

damianburrin, 2022-10-01:
Trying to flash with Windows 10
Version drop down won't poputlate

[image]

mattloretitsch-od, 2022-10-02:
Same issue on linux this morning. Might just be something upstream not related to the app?  Trying to figure it out.

garthvh, 2022-10-02:
Seems to be downloading 1.3.41 to the folder where the flasher is running and then not displaying it in the drop down on Mac

mschaavel, 2022-10-03:
Same issue on Windows 11.

F4VSJ, 2022-10-04:
I have the same issue on Debian Linux and on Win 10 Pro, tried both, was not sure
The latest firmware is downloadd (1.3.41.80ddb81) is showing up very short when clicking  "get versions" and afterwards the firmware version disappaers...
in the terminal:
 start of get_versions
 tag:v1.3.41.80ddb81 populate in dropdown?:True
 tag:v1.2.65.0adc5ce populate in dropdown?:False 
 got versions
is shown.
When triing to upload there is this in the "flashing" terminal:
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:c9:a3:fb:6c:f4
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.5s
Hard resetting via RTS pin...
Step 2/4 esp32 full
ESPTOOL Using command:-b 115200 --port COM16 write_flash 0x1000 /system-info.bin
usage: esptool write_flash [-h] [--erase-all] [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}] [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress] [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: '/system-info.bin'

maybe the hint is hidden here???
See you
hans F4VSJ

Brovka1985, 2022-10-04:
the same been going on for me for days now

garthvh, 2022-10-05:
1.3.25 is out, dropdown is fixed 
pip install --upgrade meshtastic-flasher

closed 2022-10-05 by garthvh

F4VSJ, 2022-10-05:
Yes... great Thanks for your action!
It works
Hans F4VSJ