meshtastic/Meshtastic-gui-installer #243: New relaese of Flasher and Firmware 1.3.42.xxxx

F4VSJ, 2022-10-10:
Hello all,
This morning i wanted to upload firmware to a new ttgo tbeam card, i discovered that the Flasher (in Linux) showed up that there is a new version. Great i think, I update...so done quickly , updated to version 1.3.26
The older one showed up with 2 firmwares for the ttgo tbeam: 1.3.42.xxxx and 1.3.41.xxxx
So, uploading the lates .43 firmware with the old(er) Flasher in "hang at stage 1 of 4 with an error message... don't remember but I had that already once, and after upgrading the flasher it was ok.
So upgrading the flasher with: pip install --upgrade meshtastic-flasher, and problem gone.
BUT now it is only showing the latest version of firmware in the flasher, which gives me a bit of headache:
Upload goes fine, than I set via the CLI the bluetooth active and the pairing code fix  to 000000

Know it comes, I don't find the possibilitie to connect via bluetooth to this device. I "see" the card ( in the phone app) and when clicking on it ,it simply disappears without making a connection, or asking for a pairing code

The card is running and i have to use the CLi to set the naming, channels etc etc , which works great via the meshtastics app on my phone normally
Any ideas, what can be the problem, or how can I upload the .41 firmware which works great

Kind regards from France
F4VSJ

garthvh, 2022-10-10:
The 1.3.41 version is no longer available in the flasher as the QR codes are not compatible with 1.3.42, you can download it from GitHub but you are better off getting 1.3.42 working, there is no working bluetooth functionality in either the flasher or the CLI.

1.3.43 has not been released, so I don't understand any of that.

closed 2022-10-10 by garthvh

F4VSJ, 2022-10-10:
thanks for the fast answer!
So i'm waiting the next firmware
have a nice evening
Hans F4VSJ