meshtastic/Meshtastic-gui-installer #248: "No module named meshtastic.radioconfig_pb2"

lycaon42, 2022-10-27:
When trying to start the flasher through CLI on Macbook, I receive this error:

>  Traceback (most recent call last):
>   File "/Library/Frameworks/Python.framework/Versions/3.10/bin/meshtastic-flasher", line 5, in <module>
>     from meshtastic_flasher.main import main
>   File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/meshtastic_flasher/main.py", line 11, in <module>
>     import meshtastic_flasher.form
>   File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/meshtastic_flasher/form.py", line 31, in <module>
>     from meshtastic_flasher.settings import Settings
>   File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/meshtastic_flasher/settings.py", line 11, in <module>
>     from meshtastic_flasher.position_form import PositionForm
>   File "/Library/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/meshtastic_flasher/position_form.py", line 9, in <module>
>     import meshtastic.radioconfig_pb2
> ModuleNotFoundError: No module named 'meshtastic.radioconfig_pb2'

After uninstalling the flasher, deleting the directory, creating a new directory, and updating my flasher using "pip install --upgrade meshtastic -pre" the problem persists.

caveman99, 2022-10-27:
```
pip install --upgrade meshtastic
pip install --upgrade meshtastic-flasher
```

caveman99, 2022-10-27:
radioconfig is an old 1.2 proto - if the error persists you have an old installation in your search path. Not a bug.