meshtastic/Meshtastic-gui-installer #260: Failed to start stub. Unexpected response: b'HAI'

m8st2ng, 2022-12-14:
Step 1/4 esp32 full
ESPTOOL Using command:-b 115200 --port COM3 erase_flash
esptool.py v4.2.1
Serial port COM3
Connecting...

.

.

Detecting chip type...

Unsupported detection protocol, switching and trying again...
Connecting...

.

.

Detecting chip type...

ESP32

Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 58:bf:25:05:73:e0
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.7s
Hard resetting via RTS pin...
Step 1/4 esp32 full
ESPTOOL Using command:-b 115200 --port COM3 write_flash 0x00 2.0.7.91ff7b9/firmware-tbeam-2.0.7.91ff7b9.bin
esptool.py v4.2.1
Serial port COM3
Connecting...

.

.

.

Detecting chip type...

Unsupported detection protocol, switching and trying again...
Connecting...

.

.

.

.

Detecting chip type...

ESP32

Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 58:bf:25:05:73:e0
Uploading stub...
Running stub...
Traceback (most recent call last):
File "C:\Python310\lib\site-packages\meshtastic_flasher\esptool_form.py", line 71, in run

esptool.main(command)
File "C:\Python310\lib\site-packages\esptool\__init__.py", line 684, in main

esp = esp.run_stub()
File "C:\Python310\lib\site-packages\esptool\loader.py", line 914, in run_stub

raise FatalError("Failed to start stub. Unexpected response: %s" % p)
esptool.util
.
FatalError
:
Failed to start stub. Unexpected response: b'HAI'

garthvh, 2022-12-14:
Please try the web based flasher https://flasher.meshtastic.org/

m8st2ng, 2022-12-14:
Please disregard last email. I used a different computer and it worked! Thank you.Sent from my iPhone

closed 2022-12-14 by garthvh