meshtastic/Meshtastic-gui-installer #265: File Not Found

LamonJ, 2023-02-22:
Ï got the python flasher to startup, however, every time I try to flash the firmware it says that the file doesn't exist (file not found). I have tried both versions that the flasher loads automatically and still nothing...
[image]

LamonJ, 2023-02-22:
Well, I think I bricked the RAK4631. I was thinking about using the nRF connect windows app but I didn't see any way of installing the firmware from a .zip file. Also, the Bluetooth doesn't show up now either so I can not do an OTA install, I think I am screwed. Is it even possible? I have run the meshtastic-flasher in multiple folders with multiple permissions and I always get the EXACT error. what started this problem is when I tried a bigger screen on the RAK19007 I did not notice that the VGG and GND pins were switched. I powered it on and the screen didn't seem to work so I unplugged it and put the old working (not anymore) back on and it didn't work that's when I noticed that the pins were flipped. I then took off the core (RAK4631) and put it on a RAK19003 and that's when... NOTHING works. I bought an ESP32 core but I have had NO luck at all with the ESP32, I now have tried the RAK4631 on the RAK19001 the RAK19003, and the RAK19007. The module is cheap enough but having to pay $32 in shipping, now that adds up fast. I could really use someone's help.

sachaw, 2023-02-23:
For the rak boards use dfu mode to flash (double press power button and drag & drop firmware image)
For everything else use the web flasher.

All of these methods are detailed on the website

LamonJ, 2023-02-23:
That doesn’t work I got it fSent from my iPhone

LamonJ, 2023-02-24:
Doesn’t work, well it worked once but then not again. When I double press nothing happens Sent from my iPhone

sachaw, 2023-02-25:
Ok, first, plug your RAK board into your pc, double press the power button in quick succession (it may take a few tires if you're not fast)
If no drive gets mounted check device manager to see if there are any serial ports, or see if it is detected at all using something like `lsusb`  on Linux