meshtastic/Meshtastic-gui-installer #268: I tried first the webinstaller, this succeeded, but my T3S3 v1.1 keeps rebooting, so I tried the GUI-installer to no avail

rudydevolder, 2023-03-04:
I am using the T3S3 v1.1
I must say that I must first press the boot and reset, then releasing the reset, otherwise my T3S3 keeps rebooting in less than a second.
Via windows 10 I could wipe and reinstall. But after I reset the T3S3 it keeps rebooting. It seems I must hold the Boot button while resetting. Then the port seems to be detected and I can reinstall again via the webinterface in my W10 machine. But after reset it again starts rebooting.
So I installed the GUI-installer in my Ubuntu virtual machine (VMWare 17)
VMware reports the USB port as USB JTag serial debug unit;
But when I try to flash I get this:
 **Warning: Could not find any ports using the Meshtastic python autodetection method.
 from find_ports ports:['/dev/ttyACM0']**

When I try the webinstaller via my Ubuntu virtual machine it also throws an error:
No compatible device found, when pressing info: "No port selected"

caveman99, 2023-03-04:
The GUI Installer does not support ESP32-S3 based devices. At all. There's not much development happening there, your best bet with this device is to use the Command line install scripts included with the firmware archive

Also make sure to use 2.0.23 with that board. Do you have the SX1262 variant or the one that just says LORA32?

rudydevolder, 2023-03-06:
Hi Caveman99, thanks for the reply, the board has indeed a SX1262,  
But I am also wondering why flashing via the WEB "seems" to work. It says it's erasing, uploading and then finished.
I used v2.0.23 per your advice.
After this I tried to check the LOG-file. And all I observed is these messages in the terminal:
**ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a7c
waiting for download**
Nothing more. So, I reset the board but then it starts rebooting all the time.

===================================================

GREAT NEWS:
==========

I just tried installing via the CLI. It works now!

closed 2023-03-08 by rudydevolder