meshtastic/Meshtastic-gui-installer #270: get version not working...

nis0521, 2023-03-23:
When I press the Get Versions Button. 

I get this:

start of get_versions
tag:v2.1.3.8c68d88 populate in dropdown?:False
tag:v2.1.2.6d20215 populate in dropdown?:False
got versions

And nothing in the drop down box

nis0521, 2023-03-23:
I normally use the web flasher but it’s the reason I’m using the python version. 

I was updating my firmware and the install timed out half way. 

The node doesn’t even turn on now. The board gets power judging by the constant red led. But doesn’t boot up. 

Used the python and it’s recognized but the versions won’t load. 

Can you help?

eluder87, 2023-03-26:
I have the same issue. attempted to update with the web flasher but now my device is stuck in a boot loop. I went and downloaded the python flasher but it is unable to get the versions and I get the same message as above.

rusty-labs, 2023-04-09:
WebFlasher is not working, it's showing some progress and success, but in the end I get unresponsive device with black screen
Python flasher is working, but I have the same issue, there is only current version in the dropdown list

carbofos, 2023-04-18:
Yes, the same problem. 

$ uname -a
Linux system17 6.2.0-gentoo-x86_64 #2 SMP PREEMPT_DYNAMIC Wed Mar 15 02:49:32 EET 2023 x86_64 AMD Ryzen 9 3900X 12-Core Processor AuthenticAMD GNU/Linux
```
~/Programs/Meshtastic-gui-installer-2.0.2 $ meshtastic-flasher
start of get_versions
tag:v2.1.9.d43ddc9 populate in dropdown?:False
tag:v2.1.8.ee971e3 populate in dropdown?:False
got versions
warning: queue 0x559844df0b80 destroyed while proxies still attached:
  wl_display@1 still attached
  ```

SebastianKumor, 2023-04-28:
same issue here